florianfesti/boxes #681: ontop lid incompatible with default TypeTray separators

jasondunsmore, 2024-05-30:
When an ontop lid is used with a TypeTray that has separators, the alignment pieces run into the separators.  As a possible fix, the inner height of inner walls should be made lower when ontop lid is used. Alternatively, there could be a small cutout in the tops of the separators to allow clearance for the alignment pieces.

**To Reproduce**

    boxes TypeTray --sx=100/2 --sy=100/2 --h=100 --top_edge=e --Lid_style=ontop --gripwidth=0

**Expected behavior**

The lid should fit on the box.

jasondunsmore, 2024-05-31:
[image]

florianfesti, 2024-07-16:
Yes, easily reproduced. Still meditating on what's the easiest way so solve this.

florianfesti, 2024-08-02:
OK, reduced the lips to small pieces in the corners. There may still be issues for very small (<= 4 * thickness) compartments. But I am ignoring this for now.

commit 4d5f4bfa (2024-08-02): Limit lip for "ontop" lid to corners

This avoids colliding with inner walls (if there are any).

Resolves: #681
closed 2024-08-02 by florianfesti