florianfesti/boxes #683: RackBox generator holes in mounting part not generating in correct place

unsigneduk, 2024-06-11:
**Describe the bug**
2 of the holes generated for mounting are not within the set distance entered in the parameters. 

**To Reproduce**
https://boxes.hackerspace-bamberg.de/RackBox?language=en, I use Firefox and the latest version of lightburn. 

**Expected behavior**
Generally I leave most of the settings for all the generators at the defaults in this case it is 7mm from the edge which seems to be working for all other hole positions just not the two towards the back of the mounting plate. 

**Screenshots or Drawings**
If the bug can't be reproduced easily, add screenshots or attach a drawing to help explain your problem.

**Additional context**
Add any other context about the problem here.

florianfesti, 2024-06-12:
Yeah. The issue is that the whole is located from the top of the whole box and not from the top of the front piece which is one material thickness lower. So I guess the solution is to make the front piece higher - at least on the sides.

florianfesti, 2024-06-12:
OK, there is actually a lot more wrong with the height of the box. Also the top holes should probably not be relative to the top of the box but in the grid of the mounting rails.

unsigneduk, 2024-06-12:
It looks to me with a clearer head this morning that the box is being produced acuratly just not the two holes mentioned. However as a temporary solution within Lightburn you can "ungroup" the offending part move the mounting hole a few mm and re-group. Solved at least until it is fixed.

florianfesti, 2024-06-15:
Ok, turns out this box is not really meant for a regular Rack with imperial hole spacing. So I now have the holes in the corners as the parameters suggest.

closed 2024-06-15 by florianfesti

unsigneduk, 2024-06-15:
Thanks Florian thats brilliant. Really handy script among others too.

commit 6174b32d (2024-06-15): RackBox: Fix height of front

Also fix the position of the mounting holes.

Thanks to https://github.com/unsigneduk for reporting!

Resolves: #683