florianfesti/boxes #685: Additional feature/improvement for Rounded boxes

Annatios, 2024-06-19:
Hello,

I'm a recent user for the boxes.py, and tried to do some rounded boxes, and i'd like to know if it's possible to add some features like :

- Is it possible for the inner corner of the rounded boxes to be square, as if it become possible we will ba able to add tray insid by examples.

- to facilitate the assembly is it possible to avoid having finger joint in the flex areas?

- also linked to the request above, can it be possible to add a setting let us decide wher will be the finger joint?

Maybe some of those request are already  existing and i wasn't able to use it correctly, in this case don't hesitate to tell me how to solve the problem by my self.

Have a good day

florianfesti, 2024-07-21:
Not 100% sure I understand your request. 

You can use a `radius` below two times the material `thickness` to get a square hole. Otherwise it may be easier to just adjust the top hole in Inkscape to fit the tray you want to insert.

To you remove the dove tail joints by setting `Settings for Dove Tail Joints` -> `size` to a large value.

florianfesti, 2024-08-01:
Closing due to lack of activity. Feel free to re-open if there are still questions left.