florianfesti/boxes #686: Install on MacOS Sonoma - ModuleNotFoundError: No module named 'markdown'

the4cesof, 2024-06-20:
Following the process at https://florianfesti.github.io/boxes/html/install/macos.html everything was successful until I tried to run `./scripts/boxesserver`

First, I had to change it to `./boxes/scripts/boxesserver`

- But then I get an error:
Traceback (most recent call last):
  File "/Users/4ces/boxes/./scripts/boxesserver", line 34, in <module>
    import markdown
ModuleNotFoundError: No module named 'markdown'

- checked the steps. The first time when I ran `pip3 install Markdown affine shapely`
the output was "Successfully installed Markdown-3.6 affine-2.4.0 numpy-2.0.0 shapely-2.0.4"

I ran it again just in case. this time it gives an error: 
"error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try brew install
    xyz, where xyz is the package you are trying to
    install."

So I tried `brew install Markdown` but this did not change my error.

when I run `python3 -c "help('modules')"` markdown is not in the list. Wondering if I have more than one python install here... sorry I'm not sure if this is a bug or something wrong with my specific installation.

the full text of my efforts are attached:
[terminal output.txt.txt](https://github.com/user-attachments/files/15906848/terminal.output.txt.txt)

marauder37, 2024-06-30:
You do have more than one Python install. You have system Python 12 and Homebrew Python 12.

The instructions you followed will not work for Python 12. [1]

Try this:
```
cd /Users/U3er/boxes
python3 -m venv venv
source venv/bin/activate
pip install -r requirements.txt
./scripts/boxesserver
```

This is the brief version of the instructions that pip3 spat out at the end of your terminal log.

To see which installed Python is actually in use, run `which python3`.

If it says: /Library/Frameworks/Python.framework/Versions/3.12/bin/python3 it is the macOS system python
If it says: /opt/homebrew/bin/python3 it is the Homebrew python
If it says: /Users/U3er/boxes/venv/bin/python3 it is the virtualenv that you set up using the instructions above

[1] https://docs.brew.sh/Homebrew-and-Python#pep-668-python312-and-virtual-environments

florianfesti, 2024-07-16:
Did that help? Is the command block above something that should go into the MacOS install instructions?

florianfesti, 2024-07-31:
Closing due to lack of activity. Feel free to re-open or just continue the discussion if the issue is still not resolved.

closed 2024-07-31 by florianfesti