florianfesti/boxes #687: new category for boardgame

gauthier12, 2024-06-22:
Hi,
I have written a box to store a game named "codex naturalis".
I would like to share the design and push it but I am wondering if it would not be better to create a new category "boardgame" or "boardgame organizer". I've seen that there are already a Agricola insert.
I could use to store future designs for boardgames.
What do you think about the new category ?
Cheers 
Gauthier

florianfesti, 2024-06-23:
I am all for a board game category. IIRC we even discussed that back then already. But having a category for just two generators is cluttering the UI too much. I think we need at least 3 generators and may be a forth as a perspective.

 Me finally getting #234 done might help with that...

florianfesti, 2024-07-06:
So please re-open as soon as we have a few more board games inserts. In the mean time please put your generator into Misc even if it is not that great.

closed 2024-07-06 by florianfesti