florianfesti/boxes #703: Thoughts on cardboard box generator.

ccrome, 2024-09-07:
Have you given thought to a cardboard box generator?  Something that gives cut and scribe lines for folds.

**Is your feature request related to a problem? Please describe.**
My problem -- I just need some quick and dirty cardboard boxes of a specific size.

**Describe the solution you'd like**
* Box generator with different top and bottom closures. 
* There are so many options.  Not sure where to start, but I guess just a simple box with taped bottom and a folding tuck-in top.

** Pictures of something similar **
https://www.template.net/design-templates/box-templates/sample-cardboard-box/
[image]

florianfesti, 2024-09-07:
There is the PaperBox generator. Any changes you need there?

ccrome, 2024-09-08:
🤦‍♂️ I hadn’t seen that one!  I was searching for cardboard. That looks perfect!  Thanks!

closed 2024-09-08 by ccrome