florianfesti/boxes #706: Sliding Drawer rounded outer corners

pellematrose, 2024-09-13:
The sliding drawer outputs now files with roundet outer corners instead of simply rectangular ones.
I could not find any parameters to change. Some weeks ago I could not see this behaviour.

[image]
[image]

Is it a feature or a bug?

Thank you!

florianfesti, 2024-09-13:
See https://florianfesti.github.io/boxes/html/api_burn.html and https://florianfesti.github.io/boxes/html/faq.html#why-are-there-tiny-weird-loops-in-the-corners

florianfesti, 2024-09-13:
You probably set the burn correction to something else than zero.

pellematrose, 2024-09-13:
> You probably set the burn correction to something else than zero.

You are right. I set it to 0.6. I wanted 0.06.
But thats actually a nice feature if you want rounded corners. ;)

closed 2024-09-13 by pellematrose