florianfesti/boxes #707: Flexibox (rounded box) generation with more outter walls

LadyChessica, 2024-09-16:
Hello and thanks for your hard work.
I want to know if there is a way to implement more outter walls for rounded boxes or flexiboxes.
I need one with 6 outter walls.
I tried to modify the py file, recompile it and change its name to the same as the original, but no changes appear in the web.
I don't know what I'm doing (wrong).
Thanks in advance.

florianfesti, 2024-09-17:
Funny thing. I have a RoundedRegularBox in my working dir but it turns out to still be just a plain copy of the RoundedBox. Turns out past me wasn't motivated enough to do the actual work... 

But to your question: How are you running the web instance. https://florianfesti.github.io/boxes/html/install.html#running-from-working-dir is probably the easiest way. Just start `scripts/boxesserver` and point your browser at http://localhost:8000

florianfesti, 2024-09-17:
Easiest way to implement is probably using `polygonWall` and `polygonWalls`:
https://github.com/florianfesti/boxes/blob/e901e8af81c312a0f423310d4d01442677e312de/boxes/__init__.py#L2811 https://github.com/florianfesti/boxes/blob/e901e8af81c312a0f423310d4d01442677e312de/boxes/__init__.py#L2868

The later probably needs some extension to be able to deal with zero degree "corners" and then add Dovetail Joints.

florianfesti, 2024-10-23:
OK, put something together along the lines I described above. Please test as soon as it is build into the web instance. If everything looks fine, please attache a nice picture either here or on #628

closed 2024-10-23 by florianfesti

LadyChessica, 2024-10-24:
I have take a look and it do what I need, sorry to be late.

florianfesti, 2024-10-24:
It is now available at https://boxes.hackerspace-bamberg.de/RoundedRegularBox 

You can just pick the number of corners you want.

LadyChessica, 2024-10-24:
Awesome, thanks!

commit e0676080 (2024-10-23): New generator: RoundedRegularBox

Resolves: #707