florianfesti/boxes #709: Request : CardBox V2

noscollections, 2024-09-19:
They are already a CardBox generator but an other version could be done

an exemple of a second design can be see here :
https://www.missionssecours.com/?product=support-de-cartes-pour-pioche
https://www.missionssecours.com/wp-content/uploads/2023/04/card_pioche_V04_mini_duo.png

as parameters we could have

- inside sizes for the cards
- number of locations for the cards (picture exemple have 2)
maybe we can have several lines and columns, like 2x2 , the picture is 2x1

- height if the card box (define how many cards we can add inside)
- in the midle of card location the hole. none, circle, hexagon .... size
- on the border half circle for easy card taking. non, yes, size
- border width in corners or empty border lenght for previous parameter 

this could be insert in many other boxs already on the website.

This is my idea.

florianfesti, 2024-09-19:
Did you have a look at https://boxes.hackerspace-bamberg.de/NotesHolder?language=en ?

It does not support multiple row and columns but it does allow multiple stacks in a row.

There is also https://github.com/florianfesti/boxes/pull/593. I have started adopting some of the features for the NotesHolder generator but there are still some missing.

noscollections, 2024-09-20:
Yes this is a good base.
the solid border is not very useful for handling cards when it is a draw pile that is handled a lot by players
With some option, it could do the job