florianfesti/boxes #710: House, several floors

noscollections, 2024-09-19:
That could be good to do some houses

House for decoration, for play (playmobil , lego minifigure, .... )

This exemple is great : https://3axis.co/laser-cut-textured-bakery-building-dxf-file/joed2r61/

We can stack several part

**part 1 : the roof**
we can do the roof and mybe a color on the svg could be for the roof tiles and we can adjuste the laser to draw and not cut !
Several or no sedign on the roof could be made
option a windows on the side of the roof (wall)

**floor**
we can made each floor individualy
4 walls or only 3 open on the back . children can pley inside with toy men

we can chose number and size of the windows on earch walls
maybe the wall decoration could be choose (none, deco 1 deco 2 ) cf picture for exemple)

**parameters**
we could choose sizes of roof and floors(h x p x l)

we can do a city or a town with differente sizes of house (building)

florianfesti, 2024-09-19:
I always wanted to do a castle set with walls and towers. But houses are also fine.

This is probably not too difficult to do coding wise. Overthinking the features and deciding the details is typically what makes them hard.

noscollections, 2024-09-20:
for define what to do on a wall (doors and windows)
a field can be completed for each wall. In this field we could write a D for door and W for window (and E for empty place)
WDW will do a windows a door in the middle and a second window. the easyest do is  that their centers are at equal distances.
door are set on the bottom and the windows on the middle of the height of the wall.
And we set size of door ans windows on other parameter (but all door have the same size and windows the same other size)
for decoration, not so easy, I agree.
For door and windows a specific color could be nice. We can decide with the laser software to cut them or draw them with less power lazer.

florianfesti, 2024-09-20:
Yeah, different colors for cutting and engraving is something we already do elsewhere.

I don't thing using h x p x l really cuts it. I think we need a scale. Floors might have different heights even at the same scale. And we want doors and window sills be the same height at the same scale. A wooden shed would still be lower than a story of a city villa.