florianfesti/boxes #718: ﻿The option "add_lidtopper" does not create a piece to the correct size.

nomen-niesco, 2024-11-02:
**Describe the bug**
﻿When using the "add_lidtopper" option, the created piece does not come out to the correct size. It is too long and not tall enough.

**To Reproduce**
﻿To reproduce my configuration, which is causing the issue, here's the URL of my settings:
﻿[https://boxes.hackerspace-bamberg.de/CardBox?FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.1&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&y=89&h=37.5&outside=0&sx=65*2&openingdirection=front&fingerhole=regular&fingerhole_depth=20&add_lidtopper=0&add_lidtopper=1&thickness=2.8&format=svg&tabs=0.0&qr_code=0&debug=0&labels=0&labels=1&reference=100.0&inner_corners=corner&burn=0.06&language=fr&render=0
](https://boxes.hackerspace-bamberg.de/CardBox?FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.1&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&y=89&h=37.5&outside=0&sx=65*2&openingdirection=front&fingerhole=regular&fingerhole_depth=20&add_lidtopper=0&add_lidtopper=1&thickness=2.8&format=svg&tabs=0.0&qr_code=0&debug=0&labels=0&labels=1&reference=100.0&inner_corners=corner&burn=0.06&language=fr&render=0
)

**Expected behavior**
﻿I'm waiting for the creation of a piece that makes up the width between the gaskets on the lid and the depth between the edge of the box and that of the lid (in internal dimensions).
In my case :

- Target *﻿(my measurements taken in Inkscape)*: 
  - 126.04 mm length
  -  89.12 mm width

- What i got *(measurements taken with calipers)*:
  - 138.36 mm length
  - 83.22 mm width

**Screenshots or Drawings**
﻿Here is a photo where you can see the lidtopper on the closed box and notice that the dimensions are wrong.
[image]

closed 2024-11-04 by florianfesti

nomen-niesco, 2024-11-04:
﻿Hello, have the changes been sent to the website? If so, there is still a dimension issue.
﻿﻿I've re-revised it by making a montage in Inkscape, and the piece is still too long.
﻿﻿﻿﻿Dimension 02 should be the same as Dimension 01, just like shown on the assembly diagram below:
[image]

florianfesti, 2024-11-04:
OK, this drawing confused me for a minute. The 02 dimension is not for scale, right? Right?

There was the slack missing from the lid topper in both directions. The lid is 0.2 time the thickness smaller than the box to allow movement. The lid toppers were missing that.

Might take a few minutes for this to reach the web instance

nomen-niesco, 2024-11-04:
﻿Indeed, I omitted the functional clearance at the ends of measurement 01 because I didn't know how much it should be. I preferred to leave that up to my creator/maintainer since they know what's needed.

More context about the diagram above: These are real-sized assembled parts under Inkscape. You can see that the lidtopper now has the correct width, but is still too long and should be of length "dimension 01" minus functional clearances, which I leave to my creator's discretion.

florianfesti, 2024-11-08:
I think your graphics is wrong. The lips overlap the lid only one material thickness. The fingers are actually over the side walls with are further out.

Also in the picture above the topper is about 2 thicknesses too wide. and now they are shorter by 2.2 thicknesses.

nomen-niesco, 2024-11-09:
Oh... sorry about that... my bad, *this is why i souldn't do this late at night...* Sorry for the time loss.
And so, your rigth, this effectively complete.
Thank for you for that speed of debug.

commit 3c92ab66 (2024-11-04): CardBox: Fix Lid Topper for front openingdirection

Resolves: #718