florianfesti/boxes #721: Different thickness for top and bottom fingers

dizcza, 2024-11-11:
I need to glue a 1.5mm thick PCB board to a 4mm thick wooden box designed in Boxes (previously I used OpenScad and lasercut toolbox. I like the rendering routine but the toolbox is limited and finger joins are not symmetric - that's why I'm switching to Boxes).

Assembled picture:

[image]

I just ordered a PCB with the following edge cuts:

[image]

Now I need to finish the corresponding wooden parts:

[image]

Same for left, back, and front.

I modified the `draw_finger` function of the `FingerJointEdge` just to give you an idea of what I want:

```python
class FingerJointEdge(BaseEdge, FingerJointBase):
    """Finger joint edge """
    char = 'f'
    description = "Finger Joint"
    positive = True

    draw_num = 0

    def draw_finger(self, f, h, style, positive: bool = True, firsthalf: bool = True) -> None:
        print(self.__class__, self.draw_num)
        t = self.settings.thickness
        if self.draw_num >= 3 and isinstance(self, FingerJointEdgeCounterPart):
            h = 1.5
        self.draw_num += 1
    ...
```

This solution is quick and dirty. How to do it properly?

Related to https://github.com/florianfesti/boxes/issues/109 but I don't want to revive a 5-year-old thread with no comments... Besides, I need only *top* fingers to have different thicknesses, not all / inner layers.

florianfesti, 2025-01-03:
The trick here is to create a new set of FingerJoint Edges. Just create of copy of the settings and have it create a new set with different character.

The RegularBox does something like this for the non 90° edges: https://github.com/florianfesti/boxes/blob/485164e8c89364460141c7b4709dd1c6ca0a7775/boxes/generators/regularbox.py#L86

There is no super convenient way to set the finger lengths but you can use a negative value for `extra_length` to get the right result.

dizcza, 2025-01-04:
I have achieved what I wanted not in a good manner but it works, so I'm closing it as I feel it's a rather rare feature request.

closed 2025-01-04 by dizcza