florianfesti/boxes #730: _swtemplate.py is not valid

AlexCPU, 2024-12-09:
**Describe the bug**
There's two templates in the generators folder, I presume the swtemplate is for wall mounted designs, but it seems to be out of date.

**To Reproduce**

- Copy _swtemplate.py to swtest.py
- Change classname to SWTest
- Find on the HTML dashboard
- error in console says:
`AttributeError: module 'boxes.edges' has no attribute 'SlatWallSettings'`

**Expected behavior**
Template works

florianfesti, 2024-12-09:
This is just a quick untested fix based on the code found in wallwrenchholder.py. Feel free to reopen if it does still not work properly.

closed 2024-12-09 by florianfesti

AlexCPU, 2024-12-09:
Thanks, that's pointed me enough in the right direction!

Couple of changes to actually make it work (I don't think I have the permission to reopen):

`super().__init__(self)`
goes to
`super().__init__()`

Remove the FingerJointSettings line, as that gets automatically added.

florianfesti, 2024-12-09:
Fixed. Now the `ignore-errors` is no longer needed.

commit da9bdad2 (2024-12-09): Rename _swtemplate.py to _walltemplate.py

Move to "new" boxes.walledges infrastructure

Resolves: #730