florianfesti/boxes #731: French Cleat edge stud can clash with cleat

AlexCPU, 2024-12-09:
**Describe the bug**
French Cleat settings don't ask for the height of the cleat, so can generate the stud where there isn't space on the wall

**Screenshots or Drawings**
[image]

**To Reproduce**
The French Cleat Settings requests the depth (green in fig.), and the spacing (purple in fig.), but doesn't request the height (blue in fig.) of the cleats, so this means that in stud mode (the default and likely most common), studs can be created in studs that they don't fit.

**Expected behaviour**
For holders smaller than a cleat, omit the stud.
For holders bigger than a cleat, move the stud up if it would collide with a cleat.

**Additional context**
I'll have a go at fixing this soon, but I've got a couple of generators I want to try first.

florianfesti, 2024-12-10:
Yeah, the stud implementation is pretty simple - not to say brain dead. Probably result of me not having an actual French cleat wall. A patch is very welcome!

florianfesti, 2026-10-10:
Writing code for this I wonder if these are actually the right options. Technically one only needs the stud if it falls into the gap between the first two cleats as after that the box has the second cleat as support. So may be the options should be

* none
* stud_if_needed
* always_stud
* hook_no_stud
* hook_stud_if_needed

Or have two separate options:
stud:
* never
* if_needed
* always

But hook and stud always doesn't make much sense.

Thoughts?