florianfesti/boxes #735: Add hinged lids to WallHopper

AlexCPU, 2024-12-12:
By suggesting a new box generator or other feature you agree to test cut it until it works. You understand that this may take a few attempts. As soon as it does work you will provide a sample image that can be used in the user interface or the documentation.

**Describe the solution you'd like**
WallHopper generator has optional hinged lids for the top and bottom. As proposed in #734 

** Pictures of something similar **
Basically this, but fixing the overlapping cut lines on the hinge.
[image]

florianfesti, 2024-12-31:
There are a few other tickets open against this edge already: #576 and #372

It needs to become a little more versatile to adjust to more situations - preferable including this one.

florianfesti, 2025-01-03:
OK, setting the `style` to "flush" should at least solve hopper situation. Although drawing the sides by hand allows it being even nicer.

"Flush_inset" might work with the top - although not quite as sturgy as one might want. The issue here is that the wall edges don't leave any meat on top for the hinge to connect to. A small piece E or F edge added using a CompoundEdge might solve that.

Note that I just changed the `style` business in d4b238c8d251ecc2be06fefd11ceadc10477dbac