florianfesti/boxes #736: rectangleWall and trapezoidWall do not make the same wall for the same parameters

mjhandel, 2024-12-13:
## Description
The dimensions of the wall generated by rectangleWall(h,w) is not the same as the dimensions of a wall generated by trapezoidWall(h, w, w).

## Minimum Viable Example
See  [traprectissue](https://github.com/user-attachments/files/18121399/traprect.tgz) attached - this is a generator that creates a trapezoid wall and a rectangle wall with the same parameters.

## Expected Outcome
I would expect that the two walls are the same size; the trapezoid is smaller than the rectangle. The size difference means that these two wall parts can't be used in the same design without significant effort to adjust the sizes. 
 
## Screenshots or Drawings
Image: [image]

## Additional context
* Of course, the other way to "fix" this is in a mixed-part design to only use trapezoidWall()s  

* Is the current behavior the right behavior? I don't think it is, since it does mean that you can't easily mix and match the trap walls with rect walls. 

* In a similar vein, is overallheight (__init__.py:2620) computed correctly? I think edges[2].spacing needs to be added in?

* I'll try to write the patch to fix these issues.

florianfesti, 2024-12-31:
Yes, the top edge does not have a correction for the `.startwidth()` of the edge. This is sortof on purpose. Basically as it is not quite clear how to do the correction. Or rather I wasn't prepared to think sufficiently long at the time. Looking at the case above isn't too helpful. You need to check examples with an angled wall. There are several different cases depending how much each edge is outset and what angle is used. `edgeCorner` should probably just handle that. But is currenty is not smart enough.

florianfesti, 2025-08-02:
Guess this needs to be fixed when there is an actual use case. So closing for now.

closed 2025-08-02 by florianfesti