florianfesti/boxes #739: Design looks correct in Lightburn, but laser cuts incorrectly - we're totally lost!

Cassolotl, 2024-12-28:
A member of our makerspace is really struggling to cut boxes designed on https://boxes.hackerspace-bamberg.de and we cannot work out why the laser cutter is behaving the way it is. We're completely lost! Does anyone have any thoughts or guidance? Any clues at all would be helpful.

The design looks correct in Lightburn, and when they click preview it looks correct too. Then they send the design to the laser cutter, and the laser moves correctly at first - and then it will randomly veer off diagonally, or recut some tabs but slightly offset, etc. They tried putting each shape into a different layer in Lightburn, and it didn't help. They've wasted so much wood that they bought specially for this project!

Can you tell us anything that might help? Would you guess that it's Lightburn, the box generator, or the laser cutter itself?

Here's some photos of some of the errors they've experienced:

[image]
[image]
[image]

florianfesti, 2024-12-30:
I have never seen anything like this. I'd guess this is an issue between Lightburn and your laser cutter. May be the data got corrupted?  On a SD Card may be?

While I don't think looking at this from an Boxes.py POV will lead to much, to do that I need the file attached here or at least the link you used to generate the file.

Cassolotl, 2024-12-31:
Here's the SVG:

[image]

I am up for sending the Lightburn file, but Github won't let me upload it because it's a disallowed filetype, so here it is on Google Drive: https://drive.google.com/file/d/16dsg4GL3-4G7rrq4ew2_FobDixzCXt90/view

> I'd guess this is an issue between Lightburn and your laser cutter.

Thank you, I have now emailed the people we bought the laser cutter from, to see if they can shed any light!

Cassolotl, 2025-01-21:
Issue resolved! It's because the member was cutting the file directly from Lightburn on the laptop, rather than sending the file to the laser cutter with the Lightburn "send" button (via USB) and then pressing "start" on the laser cutter itself.

closed 2025-01-21 by Cassolotl