florianfesti/boxes #744: Feature request: tagging releases

FedericoCeratto, 2025-01-08:
Hello and thanks for developing boxes.py 
Would you consider tagging releases when the time is right?
It would help with packaging and redistributing boxes.py as a library or local application.

florianfesti, 2025-01-11:
Sure. Currently the master branch is instantly deployed on the web instance. So we'd probably want some automation to build the releases, too.

Help for this is very welcome -  especially for making the Inkscape part work out of the box.

What release cycles do you think would make sense? Any preferences for a particular versioning / tagging scheme?

FedericoCeratto, 2025-01-26:
@florianfesti IMHO I would start with tagging the whole repository as `0.1.0` and let any distribution that wants to package ship it as an application or application+library or do their own build for the inkscape plugin if they want. Then if you'll ever feel the need to split it into component with independent versions you can do it before version `1.0.0`

florianfesti, 2025-02-10:
OK, tagged version 1.2 as we were already at 1.1

closed 2025-02-10 by florianfesti