florianfesti/boxes #745: TrayLayout `TypeError: expected str, bytes or os.PathLike object`

sethfischer, 2025-01-09:
## Description

Creating a `TrayLayout` on the command line results in the following error:

```
TypeError: expected str, bytes or os.PathLike object, not TextIOWrapper
```

## To reproduce

Python: 3.11.11
Boxes: 36e1db1c21aa65749253470da021977f7a12e5a

```console
$ cat layout.txt 
 ,> 50.00mm
 | ,> 50.00mm
 | | ,> 50.00mm
+-+-+-+
| | | |  50.00mm
+-+-+-+
| | | |  50.00mm
+-+-+-+
| | | |  50.00mm
+-+-+-+
```

```console
$ boxes TrayLayout --sx=50*3 --sy=50*3 --input=layout.txt
Traceback (most recent call last):
  File "/home/user/.cache/pypoetry/virtualenvs/boxes-builds-tv7Y1cRB-py3.11/bin/boxes", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/.cache/pypoetry/virtualenvs/boxes-builds-tv7Y1cRB-py3.11/lib/python3.11/site-packages/boxes/scripts/boxes_main.py", line 150, in main
    run_generator(name, sys.argv[2:])
  File "/home/user/.cache/pypoetry/virtualenvs/boxes-builds-tv7Y1cRB-py3.11/lib/python3.11/site-packages/boxes/scripts/boxes_main.py", line 93, in run_generator
    box.render()
  File "/home/user/.cache/pypoetry/virtualenvs/boxes-builds-tv7Y1cRB-py3.11/lib/python3.11/site-packages/boxes/generators/traylayout.py", line 476, in render
    self.prepare()
  File "/homeuser/.cache/pypoetry/virtualenvs/boxes-builds-tv7Y1cRB-py3.11/lib/python3.11/site-packages/boxes/generators/traylayout.py", line 164, in prepare
    with open(self.input) as f:
         ^^^^^^^^^^^^^^^^
TypeError: expected str, bytes or os.PathLike object, not TextIOWrapper
```

## Expected behaviour

It is expected that the command `boxes TrayLayout --sx=50*3 --sy=50*3 --input=layout.txt` executes without error.

jasondunsmore, 2025-01-09:
I see this too. If I revert commit `04656e18a`, it works from the command line and web UI.

florianfesti, 2025-01-09:
No idea what I was thinking. May be there is a bit too much magic happening there.

commit d83609ef (2025-01-09): Revert "TrayLayout: Actually read the input file"

This reverts commit 04656e18a56abe25de4e9809e0a17294baac9e1e.

Turns out things were already working just fine.

Resolves: #745
closed 2025-01-09 by florianfesti