florianfesti/boxes #753: Settings Export with SVG

DBVCA, 2025-01-31:
It would be great to add a check box option to export the box creators settings that were used in the svg file in text at the bottom of the file.  I often will want to go back to previous files I have and want to make small tweaks or alternate versions.  It would be such a time savor for all the settings used for that creation, so they can be referenced at a later time.

gcollic, 2025-02-01:
Hello,

Just to add some context to the conversation, it's already possible to have access to this references afterward.

Here is how I personally do this:

SVG files are text files. If you open the file directly in a text editor (notepad, VS Code, etc.), you can see there is some comments at the start of the file, containing the box type and it's parameters.

It's definitely not as obvious and discoverable as a check box generating a visible text box (but it also doesn't need to be deleted/excluded before cutting).

florianfesti, 2025-02-01:
The settings are also available in the SVG file's metadata. E.g. in Inkscape you can go to `File` -> `Document Properties` and then switch to the `Metadata` tab. Other programs should have similar functionality.

florianfesti, 2025-02-01:
This is even documented [in the FAQ](https://florianfesti.github.io/boxes/html/faq.html#what-settings-were-used-to-generate-a-drawing)

florianfesti, 2025-02-03:
Basically I think we already have enough way to get the original settings. There is also the `Save to URL` and `QR Code` buttons. So I am closing this here as WONTFIX. May be the metadata could be pointed at more prominently than the FAQ. Any suggestions?

closed 2025-02-03 by florianfesti

DBVCA, 2025-02-03:
I appeciate the response and this will be very helpful to go back to my old designs.  Thank you.