florianfesti/boxes #755: Sphereoid lamp

JolienF, 2025-02-01:
Hey,

So I made this lampshade (its not perfect yet, but the prototypes are looking good). Its a sphere like shape that can be cut from flexible material like cardboard or plastic sheets. The amount of gores (https://en.wikipedia.org/wiki/Gore_(segment)) is variable. The pattern is not generated (yet). 

So my question is would this be a project to put on boxes.py? It's not the  standard wooden boxy type thing. But it is generated and very lasercuttable
I'd love to give it a go to integrate it in this codebase, but I could definitely use some tips. 

[image]

[image]

JolienF, 2025-02-01:
[image] this is the type of pattern it generates

florianfesti, 2025-02-03:
Yes, of course. There are a lot of things in there already that are not quite boxes.

ccrome, 2025-02-03:
Love it!  I'm happy to give it a try!

florianfesti, 2025-02-03:
A little more gore, what's not to like...?

JolienF, 2025-02-04:
Alright, I'll get started then!

JolienF, 2025-02-04:
> Love it! I'm happy to give it a try!

I'd like to try myself first, if you don't mind. I do appreciate some help though

florianfesti, 2025-02-04:
Generally Boxes.py is not that suited for rounded shapes. I does offer very little geometry support. The gears module uses `drawPoints()` which does kerf correction and draws a polyline from coordinates instead of lengths and turns.

An alternative could be using `curveTo` which draws Bezier curves but looks like it is missing burn correction although it is in a place in the API where it should actually do that. If this gets an issue I can fix that, though.

JolienF, 2025-03-01:
@florianfesti what do you reccomend for circles? I use some halfcircles, but the disc and circles in boxes don't have a kerf or an option to create only half a circle.

florianfesti, 2025-03-01:
You can just use `.corner` and `.polyline`. See https://florianfesti.github.io/boxes/html/api_burn.html

https://florianfesti.github.io/boxes/html/boxes.html#boxes.Boxes.polyline

https://florianfesti.github.io/boxes/html/boxes.html#boxes.Boxes.corner