florianfesti/boxes #756: Different Parameters for Type Tray

JayPerez1, 2025-02-08:
I apologize if this already exist, but I couldn't find an existing template with the parameters I'm requesting.

My Grandson has a lot of Hot Wheel toy cars that I'd like to design a storage box for. The cars are fairly consistent in size at about 1" wide by 3" long and 1" high.

It would be very similar to your Type Tray. However, it would be much easier if I could tell it the inside dimensions of the pockets and how many I want in X and how many in Y.

I drew up a rough CAD model and created an image and a drawing to illustrate and have attached them to this ticket.

In this example I would say I want the pockets to be 1.125" by 3.125" and 5 across and 3 tall. The inner dividers are 3/4" tall and the outer walls are 1" tall.

[Hot Wheels Storage Drawing.pdf](https://github.com/user-attachments/files/18716835/Hot.Wheels.Storage.Drawing.pdf)
[image]

An enhanced version might allow cutouts on the sides of each pocket to allow your fingers room to grab a car as shown in version 2.

[Hot Wheels Storage Drawing v2.pdf](https://github.com/user-attachments/files/18716890/Hot.Wheels.Storage.Drawing.v2.pdf)
[image]

I hope my descriptions are clear. If you already have such a template I would appreciate it if you could point me to it.

Thanks in advance !

Jay Perez

florianfesti, 2025-02-08:
https://boxes.hackerspace-bamberg.de/TypeTray?label_text=&sx=28.575*6&sy=79.375*3&h=25.4&hi=19.05&outside=0&labels=0

Creates a Type Tray of the size you want. You will still need to adjust material `thickness` and `burn`.

See https://florianfesti.github.io/boxes/html/usermanual.html

Generally multiplying inch measurements by 25.4 will get you the size in Millimeter - which is the default unit for technical drawings and also Boxes.py.

Un-checking the `outside` box will use the sizes as inside measurements of the pockets.

Adding cutouts for gripping is a great feature. But if you don't want to program that yourself or want for me to get to that you can just use a vector graphics program like Inkscape to just alter the pieces by hand. Create a shape of the cut out then duplicate or clone it and then subtract from the inner wall piece.

JayPerez1, 2025-02-08:
Thank you for the quick response and the information Florian !!!

Good Grief, I've been using those parameters wrong !!!

For most of my stuff I was entering the *outer* dimensions of the box I
wanted and then *dividing* it by the number pockets I wanted !  (Which
works great when you're restricted to specific outer dimensions and aren't
concerned about the exact sizes of the pockets, BTW.)

I'm able to combine my CAD and Lightburn skills in a way to achieve the
cutouts but unfortunately am not able to program, even with AI's help.

I've looked at your page for contributing $$$ but don't have a PayPal
account and am a little uncomfortable with some of the other methods from
here in the US. I'm willing to donate $100 dollars towards your great work
if we can figure out a way to get it to you.

Thanks again.

Jay Perez

florianfesti, 2025-02-09:
The PayPal link should also work without an account. You can then pay with a Debit or Credit card.

JayPerez1, 2025-02-09:
Thanks for that info Florian, I wasn't aware of that.

Just contributed 100 euros.

Jay

florianfesti, 2025-02-10:
Thank you! Very generous!

I updated the "Giving back" page to make this option a bit more obvious.

JayPerez1, 2025-02-11:
Glad to contribute. My 84 year old sister uses your tool to create all
kinds of boxes for her crafting and many members of our Rochester
Makerspace also use it.

Just went to this link and don't see how to contribute money. Should it be
there ?

boxes/CONTRIBUTING.rst at master · florianfesti/boxes
<https://github.com/florianfesti/boxes/blob/master/CONTRIBUTING.rst>

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

florianfesti, 2025-02-11:
Yes, yes it should. It shows that I just added the `Giving back` page recently. Added the link with 43e7d3df089c4ddc7769452c0e0a759eb2b63694

florianfesti, 2025-02-23:
Ok, looks like all questions got answered here. Closing.

closed 2025-02-23 by florianfesti