florianfesti/boxes #765: Cutouts for overlapping miniatures

primetime34, 2025-03-06:
**Describe the solution you'd like**
A way to create overlapping miniature holders as shown in the accompanying picture

[https://i.etsystatic.com/13033371/r/il/ba6d74/3991412743/il_1588xN.3991412743_6q8g.jpg](url)

ffesti, 2025-03-06:
Easiest way is probably to just use a [TypeTray](https://boxes.hackerspace-bamberg.de/TypeTray?label_text=&sx=250&sy=5%3A50%3A5%3A5%3A50%3A5%3A5%3A50%3A5&h=50&hi=45&outside=0) and add the cutouts to the walls with Inkscape.

florianfesti, 2025-03-12:
This is one of the use cases where it is just not worth creating a specific generator. There are so many ways one might want these slots. Just add them yourself. The link above should give you an idea on how to use the TypeTray for this.