florianfesti/boxes #766: Finger Joint Depth NOT correct on several Boxes

JohnFOBrien, 2025-03-10:
I have been attempting to laser cut some simple boxes designed on boxes.py. I am using 3mm mdf and have set the default thickness setting to 3.07. However, several of the boxes I tried this on generated finger joints that are only 1.4mm deep instead of (1 x 3.07 =) 3.07mm. I have set all the parameters accordingly as shown in the screenshot.
I will also attach the SVG file that is generated from these settings as well as the URL.
I found this issue on the Console Box as well as the Console2 box. I verified that the finger joint tabs ARE the correct depth (not width) in the Abox and BasedBox designs. That is all I have tested but the generators on the other boxes may also be providing incorrect finger joint sizes based on the thickness of the material.
I am also attaching a photo of the finger tabs generated from the console box (smaller height tabs) and the Burn Test tabs which are the correct tab height.
I hope this fully documents the issue and you can reproduce it. If you need further information, let me know.

John OBrien

[image]

[image]

[image]

florianfesti, 2025-03-10:
The SVG above does have the right size for the fingers. May be you resized it somewhere on the way to the laser cutter. That's what the reference is for. Please set it back to 100 and check the size of the reference rectangle before (or after) cutting.

The fingers are all generated by the same code for all generators. I am not saying it is impossible that there is a bug but it is very unlikely.

JohnFOBrien, 2025-03-11:
I am not sure how you verified that the SVG has the right sizes, as I have
determined that all the pieces rendered are only about half size, not
exactly but around that size. I opened the console13 SVG file in XCS laser
software. I measured the 100 mm reference rectangle at
51m. I also measured the finger as shown in the attached screen shots. The
finger width and space are suppose to be 2 times the thickness (3.07) but
instead they measure 3.25mm which is close to just 1 thickness, not 2.
Also, the finger depth or height should be just 1 thickness or 3.07MM and
from the screenshot you can see it is 1.44 mm, a little less than half of
what the box software designed it at.

The only other factor I changed in this console box is the front. I am
converting this console box into a Smart Phone Amplifier, so I am closing
the box at the apex of the triangle by setting the front size to "0" To see
if this is what is throwing off the conversions, I will generate the same
console with the front having a size of 25. I will then import the SVG into
XCS and measure the reference box and finger tabs again to see if the front
size had anything to do with it.......
##########################################################################################################
So I changed the size of the Front from 0 to 25 mm and ran through the same
tests as I noted above. I took snapshots of everything like I did for the
slanted console box but nothing changed. Everything is still undersized. I
have not included any of the photos so as to not clutter up this diagnosis.

So my last test is to prove that that a GOOD box file does indeed measure
correctly when I open that SVG file in my XCS laser software and measure
accordingly. I kept all the settings the same except for the size of the
box which is 100 x 100 x 25 mm. The design was called "Abox". I generated
the SVG file and opened it in XCS laser software. Sure enough, the
reference box is the same size as the sides (100mm) and the tabs ore 3mm
deep x 6 mm wide as they should be.
I am attaching the console settings, svg file, and screen shots of the
measurements just as I did for the console box for you to test out and see
what is going on.
Conclusions do far: The console Box renders the parts much smaller than
they are designed to be. The reference box is only 60 mm long when it
should be 100.
Using the same settings and material thickness for the Abox, the parts
rendered from the SVG file are the correct size. The reference box is
indeed 100 mm long and the measurements of the fingers are correct for the
thickness of the material.
So, if the input settings are the same for both boxes, why does one box
come out correct and the other box undersized in all dimensions?
I hope you can find a simple solution!

JohnFOBrien, 2025-03-11:
[image]
[image]
[image]

[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]

florianfesti, 2025-03-11:
Well, I downloaded the file and opened it with Inkscape. It looks just fine there. Looking at the sources the difference in the coordinates between the H commands and V commands in the paths also look right at the first glance.

Unfortunately not all tool chain the the size right. There even is an FAQ entry for that: https://florianfesti.github.io/boxes/html/faq.html#why-are-my-parts-in-the-totally-wrong-size

I wonder if XCS laser scales the drawing to fit on the workspace of your laser cutter. May be there is some setting somewhere to switch this behavior off. I don't have any experience with XCS specifically but laser software getting the size wrong is not unheard of. I added the reference for a reason.

closed 2025-03-12 by florianfesti