florianfesti/boxes #767: Add option for "Skadis" to wall mounted boxes.

sergiomiracco, 2025-03-11:
By suggesting a new box generator or other feature you agree to test cut it until it works. You understand that this may take a few attempts. As soon as it does work you will provide a sample image that can be used in the user interface or the documentation.
I'd love to do that :-)

**Describe the solution you'd like**
As you already have the _SkadisBoard_ generator, it would be nice to have a _Skadis_ option for generators in the _WallType_ section. The hook design is straight forward and should be added to the side walls. I guess the concept may be similar to the _slatwall_ option.

florianfesti, 2025-03-16:
OK, here is the first version. This will need proper testing - preferably with an original IKEA board and a lasercut one. The original has rounded edges everywhere which we obviously can't do with the laser. So the fit may vary. It also will depend on the material used.

There is also some documentation needed on how to select widths that actually work with the Skadis board.

florianfesti, 2025-04-23:
OK, I consider this done for now. Some more documentation surely wouldn't hurt. E.g. one page / chapter for Generator Groups that deserve them; like WallMounted and Flex. But may be this is just beyond the scope of this ticket. 

SkadisBoard has now an explanation on how to use the other generators. I moved it to the WallMounted group to have it nearby.

Testing is welcome. May be you can donate a picture or two of the results. They could look really nice in the descriptions of both the SkadisBoard generator or the generators of the wall mounted boxes/holders.

closed 2025-04-23 by florianfesti