florianfesti/boxes #769: Clarification of rectangularHole

shenkers, 2025-03-16:
I'm new to boxes.py and excited to learn this cool framework, thank you for making it available!

I'm trying to understand how I should be using the rectangularHole to make a centered 6x6 hole inside a 16x16 wall with fingers on the edges. 

If I use this code the hole is not centered inside of the wall, it is shifted to the lower-left:

```
ox = self.adjustSize(16)

# draw hole with center at position (8,8)
self.rectangularHole(ox/2, ox/2, 6, 6)
# draw outer wall with fingers
self.rectangularWall(ox, ox, "FFFF")
```

I have found I can draw a rectangle at the desired position using this code: 

```
ox = self.adjustSize(16)
iw = self.adjustSize(6)

# draw hole
self.moveTo(5,5)
self.rectangularWall(iw, iw, "EEEE")
self.moveTo(-5,-5)

# draw outer wall with fingers
self.rectangularWall(ox, ox, "FFFF")
```

but I feel I must not be using the API correctly, since the burn offset is set as though it is an outer wall. How can holes be positioned relative to rectangular walls? I've tried fiddling with moving the hole around by factors proportional to `thickness` but feel like I'm shooting in the dark. 

Is there somewhere to read more about the offsets applied/in what situation adjustSize needs to be used? I've been looking through the generators and documentation to try to understand, but feel like I need a programmer's guide to the library.

shenkers, 2025-03-16:
For context, these are pictures of what I am trying to achieve:

[image]

and the result of the code using rectangularHole:

[image]

florianfesti, 2025-03-16:
You want to use a callback. See https://florianfesti.github.io/boxes/html/api_parts.html#the-callback-parameter and https://florianfesti.github.io/boxes/html/_modules/boxes/generators/nemamount.html#NemaMount as a simple but not completely trivial example.

shenkers, 2025-03-16:
You are the best, thank you for the tip, I'll dig in there

closed 2025-03-16 by shenkers

shenkers, 2025-03-16:
For posterity's sake, this code does what I was looking for:

```
cb = lambda: self.rectangularHole( ox/2, ox/2, 6, 6 )
self.rectangularWall(ox, ox, "FFFF", callback=[cb], move='right')
```

[image]