florianfesti/boxes #770: Bug in Notesholder with 4.0mm thickness

marcustrp, 2025-03-22:
**Describe the bug**
Two pieces are malformed when changing thickness of material form 3.0 to 4.0mm

**To Reproduce**
[Link to boxes.hackerspace-bamberg.de page](https://boxes.hackerspace-bamberg.de/NotesHolder?FingerJoint_style=rectangular&FingerJoint_surroundingspaces=1.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_bottom_stabilizers=0.0&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&sx=78*1&y=78&h=35&bottom_edge=s&opening=40&back_openings=0&thickness=4&format=lbrn2&tabs=0.0&qr_code=0&debug=0&labels=0&labels=1&reference=100.0&inner_corners=loop&burn=0.1&language=en&render=2)

**Expected behavior**
The small bottom two parts should not have 

**Screenshots or Drawings**
[image]

aramperez, 2025-03-28:
I was just going to file this error too: https://boxes.hackerspace-bamberg.de/NotesHolder?FingerJoint_style=rectangular&FingerJoint_surroundingspaces=1.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_bottom_stabilizers=0.0&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&sx=95*1&y=95&h=51&bottom_edge=s&opening=40&back_openings=0&back_openings=1&thickness=5&format=svg&tabs=0.0&qr_code=0&debug=0&labels=0&labels=1&reference=0&inner_corners=corner&burn=0.1&language=en&render=1

florianfesti, 2025-04-09:
Yeah, there isn't enough space there for the feet. You can make them smaller by changing `width` in the `Settings for Stackable Edges`:

https://boxes.hackerspace-bamberg.de/NotesHolder?FingerJoint_style=rectangular&FingerJoint_surroundingspaces=1.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_bottom_stabilizers=0.0&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&sx=95*1&y=95&h=51&bottom_edge=s&opening=40&back_openings=0&back_openings=1&thickness=5&format=svg&tabs=0.0&qr_code=0&debug=0&labels=0&labels=1&reference=0&inner_corners=corner&burn=0.1&language=en

Increasing `angle` will also save space to some extent.

florianfesti, 2025-04-09:
Guess this does belongs in the generator itself and not in some random ticket no-one is going to see soon.

commit 49e3e9cf (2025-04-09): NotesHolder: Add description

Give hint about making the stackable feet smaller if necessary.

Resolves: #770