florianfesti/boxes #778: Corners are loops

beauxjames, 2025-04-04:
**Describe the bug**
At corners/intersections, the lines are looping vs a straight corner. This creates implications with other features as well, but this example includes a close-up of a 90 deg intersection. On master branch.

**To Reproduce**
The example is limited to the Slanted Tray, however I have verified this issue exists with other files. 
Just zoom in to any intersection.

http://localhost:8000/SlantedTray?FingerJoint_style=barbs&FingerJoint_surroundingspaces=2.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=4&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1&sx=40*4&y=110.0&h=40.0&outside=0&front_height=0.3&thickness=3.0&format=svg&tabs=0.0&qr_code=0&debug=0&labels=0&labels=1&reference=100.0&inner_corners=loop&burn=0.1&language=en&render=0

**Expected behavior**
When lines merge, there should be no spline artifacts created, causing a loop to exist. The anchor points should be fixed.

**Screenshots or Drawings**

[image]

[image]

florianfesti, 2025-04-04:
This is expected behavior. See  https://florianfesti.github.io/boxes/html/faq.html#why-are-there-tiny-weird-loops-in-the-corners and https://florianfesti.github.io/boxes/html/faq.html#i-really-don-t-want-those-weird-tiny-loops

closed 2025-04-04 by florianfesti

beauxjames, 2025-04-04:
Ha. Thank you. That really seemed like an issue, as I cannot imagine why someone would want to process corners as loops.