florianfesti/boxes #779: Does not display correctly

missaelabreugrillo, 2025-04-04:
**Describe the bug**
I'm using boxeserver and it doesn't show the images and the website doesn't load properly.

**Screenshots or Drawings**

[image]

florianfesti, 2025-04-04:
Looks like the /static folder is not being found. Try `scripts/boxesserver` instead of `boxes/scripts/boxesserver.py`. You can also use `--static_url=https://florianfesti.github.io/boxes/static` to use the files from he web instance.

Currently the boxesserver look for the static files at `../static` which does not work for `boxes/scripts/boxesserver.py`. I'll fix that soon.

missaelabreugrillo, 2025-04-04:
if I use python ./scripts/boxesserver

It throws me the following error:

File "C:\Users\Administrador\Desktop\boxes\scripts\boxesserver", line 1
    ../boxes/scripts/boxesserver.py
    ^
SyntaxError: invalid syntax

Note: I have tested this on Windows 11 and also on Linux (Ubuntu) and the same error occurs.

florianfesti, 2025-04-04:
Yes, ./scripts/boxesserver is a symlink. It should work fine on Linux but not if your checkout is on a Windows file system. Anyway, should be fixed now.

commit 625c9291 (2025-04-04): boxesserver: Be smarter about static/ on disk

Add --static_path cli option to allow passing the actual location of the
static content.

Also check ../static/ and ../../static from the location of the
boxesserver binary.

Resolves: #779
closed 2025-04-04 by florianfesti