florianfesti/boxes #803: Edges with TrayLayout (no floor) have unnecessary notch

BRFud, 2025-06-04:
Hi. I'm trying to use TrayLayout to generate a set of tray dividers with no floor, and it's putting notches at the bottom of the vertical walls. I need the bottom edges to be straight without any notches - is there some way I can achieve this? I've tried editing the script myself without much success at removing these notches properly.

[image]

Image is attached showing the notches that seem to serve no purpose.

Any advice appreciated.

florianfesti, 2025-06-04:
Yeah, this generator is a bit too complicated for it's own good. There is also  #611 still open. Not that this will affect you.

Update should be in the web instance in a few minutes.

BRFud, 2025-06-05:
Thanks for having a look at this! I thought your patch was working fine but on close inspection when I zoom right into the resulting .svg where the notches were I see that the notch has been replaced with a broken line, whereas I believe it should be a continuous joined/closed line. In my case this causes the fill to be misinterpreted when trying to extrude for 3d printing and I'm sure it would also cause issues with incomplete laser cutting, etc.

see attached image below
[image]

BRFud, 2025-06-05:
nevermind, it seems those gaps were created when I used the  --tabs option, please disregard my last message :)

commit a0456e92 (2025-06-04): TrayLayout: Omit notches if there is no floor

If walls cross with no floor in all surrounding quadrants we don't need
to leave space for that floor underneath the crossing.

Abuse "S" edge to signify to use a E edge instead of e.

Thanks for https://github.com/BRFud for reporting!

Resolves: #803